ANYCUBIC-3D/AnycubicSlicer #3: WipeTower

Locutus1988, 2024-07-04:
Hallo,

der Kobra 3 hat doch einen Reinigungsturm der die Filament Reste entfernt und dann von Drucker wegkatapultiert. Wieso wird dieser nicht während eines Mehrfarbendrucks verwendet sondern ein Turm aufgehäuft. Abfall ist Abfall aber muss doch nicht sein.
Dann hätte man sich diesen auch sparen können.

Sollte man drüber nachdenken ob sowas nicht geändert wird.
Übersetzung für Deutsch wäre auch von Vorteil.

Eine Funktion um auf einem Objekt Text anlegen zu können habe ich auch nicht gefunden. Andere Slicer können das. Leider ist mit denen der Mehrfarbendruck über ACE nicht möglich...

Für die Nächste Firmware würde ich mir wünschen, dass der Drucker nach Beendigung des Drucks das Model nach vorne befördert als Standard Einstellung. Somit lassen sich die Modelle viel leichter Entfernen.

Hello,

The Kobra 3 has a cleaning tower that removes the filament residue and then catapults it away from the printer. Why is this not used during multi-color printing instead of piling up a tower? Waste is waste but it doesn't have to be that way.

Then it could have been saved.

You should think about whether something like this should be changed.

A translation for German would also be an advantage.

I haven't found a function to create text on an object either. Other slicers can do that. Unfortunately, multi-color printing via ACE is not possible with them...

For the next firmware, I would like the printer to move the model forward as a standard setting after printing is finished. This makes it much easier to remove the models.

ljschmitt, 2024-07-06:
+1 Push

i900gaming, 2024-07-19:
+1 Push

majo1604, 2024-09-16:
Und ein weiterer Vorschlag für ein nächstes Update... warum muss der Mainboardlüfter ständig laufen, auch wenn nicht gerechnet wird? Ich weiss, dass es bei den meisten Druckern so ist. Aber es gibt auch welche, die sind einfach mal stumm im standby.

rabbit3dcustom, 2024-12-19:
The wipe tower is not working as spected... the filament prime mm3 is not working. I you compare the wipe tower with orcaslicer, or the anycubicPrusaSlicer customization the wipe towers are soooooo muuuuch little, without wasting the huge amount of filament of the "Next" version.