ANYCUBIC-3D/AnycubicSlicer #6: login

3DPapa, 2024-07-11:
Purchased and set up Kobra K3. Installed slicer and remote app on android phone. Cannot login at opening slicer so allow transfer of files or remote control. Keeps asking for phone number of 11 numbers, which I did. Will not accept 10 digit for US use. I do have an active account on Anycubic with name PapaMaker, email rbharett@gmail.com

hone89, 2024-07-14:
Hola amigo,tengo el mismo problema,encontraste la solucion?

Stumpy-finger, 2024-07-23:
I am currently running version 1.4.4 and all I get is a white popup that has nothing on it. I have tried multiple times to login over multiple days with no luck. Is there a fix for this?

PPAC37, 2024-07-23:
> Keeps asking for phone number of 11 numbers, which I did. Will not accept 10 digit for US use.
@3DPapa 
Hi! I think (but need to be checked) it is necessary to add the international designator code at the beginning of the number. If I'm not mistaken, it's a "+1" for the USA to add at the beginning of the telephone number if you're calling from a foreign country.

PPAC37, 2024-07-23:
@Stumpy-finger 

Hi !
Can you add details? (Like the operating system and its version? And a screenshot to illustrate? (yes I know it won't do much but it's just to show the problem clearly))

Because for me (Windows 10, x64, build 195045.4651) it seems to work correctly.
Under AnycubicSlicer v1.4.4, when I click on the connection icon, 

[image]

there is a window that opens (first filled with white, then there is a loading animation and finally it shows the login form. 

[image]
[image]
[image]

)

Stumpy-finger, 2024-07-23:
After I put in the comment I went back and look at my computer enclosed
Firefox then I was able to log in so if something is clashing between
Firefox and anycubic slicer