ANYCUBIC-3D/AnycubicSlicer #13: Linux version

ErikWegner, 2024-08-18:
Please provide a Linux version (e.g. flatpak).

diefans, 2024-09-01:
I run NixOS and use [PrusaSlicer](https://github.com/prusa3d/PrusaSlicer) together with [UVTools](https://github.com/sn4k3/UVtools) for my Anycubic Photon Mono M5s. Sometimes (still wondering if the firmware is the reason - but there also no downloads/updates available) when modifying the `p5ms` file I get insane high (>100 hours) print times and the printer is just stuck - so it would be nice to have a tool, which is tailored to this printer... I would not mind to compile it by myself, an AppImage would also fit...

majo1604, 2024-09-16:
Where are the Linux Version?
Or source code, so we can compile an own Version. I think, here so many Linux Users.

blurayne, 2025-02-17:
Just thought about buying Kobra S1. Now that I see proprietary c**p and Linux version = equals no option. 

Sorry, I don't run Windows and I'm not a gamer. Please provide a Linux version. At least an AppImage or Flatpak. For whatever can be closed source as far as I can run it natively on my system and don't need a VM.

brosskgm, 2025-02-27:
I even sent them an email asking about Linux version since I don't have Windoze of Mac either or at least the printer settings so we could configure the slicer ourselves. Doesn't look like it's going to happen. Looks like I won't be getting the combo either.