ANYCUBIC-3D/AnycubicSlicer #15: Other Brand Filament

joshs85, 2024-10-24:
The slicer only has options for Anycubic filament.  Please add profiles for other brands.

nolantern, 2024-12-21:
You can add as much user presets as you want.
Just edit an existing filament and save it with a new name. Keep in mind, that the new filament inherits all settings from the source filament that are not changed. Changing the source filament results changes the child filament too.

Alternatively you could import filaments created in orca slicer. Or you export some filament, edit the text file and import it as a new filament.

joshs85, 2024-12-21:
Right.  This is the problem.  They should be there already.  I don't want to have to add them myself.