ANYCUBIC-3D/AnycubicSlicer #22: Where is the source code?

dlannan, 2025-03-05:
I recently bought a Kobra S1 and I use Linux, but there is no Linux slicer version for you build.
The repo here is empty, where is the source code so I can build it for Linux?
Im quite puzzled that you are using OSS and not abiding by the license agreements by providing it?
Where can I find it?

stylon, 2025-03-20:
I have the exact same issue. Kobra S1 and only Linux as host available.
Please provide the source code.

angry-elf, 2025-03-23:
Same for me. Official support says 'somewhere in the future.' Does anybody know a lawyer who wants to make easy money?

[image]

mosfet80, 2025-03-27:
https://anycubic.it/pages/firmware-software say 
Anycubic Slicer is an open-source, easy-to-use slicing software for FDM 3D printing, based on PrusaSlicer by Prusa Research, which is from Slic3r by Alessandro Ranellucci and the RepRap community.

dlannan, 2025-03-27:
> https://anycubic.it/pages/firmware-software say Anycubic Slicer is an open-source, easy-to-use slicing software for FDM 3D printing, based on PrusaSlicer by Prusa Research, which is from Slic3r by Alessandro Ranellucci and the RepRap community.

I think you havent read properly. We are using Linux, and because we have the new Kobra S1, we cannot connect to the printer over network because it is not supported in any of the Linux slicing software. 

AnyCubic Slicer uses open source software that is actually Orca Slicer (it is _not_ their own software) and this specific software is AGPL3.0 licensed - which is a GPL based license.  To be clear, from the gnu.org page themselves:

The GNU Affero General Public License is a free, copyleft license published by the Free Software Foundation in November 2007, and based on the GNU GPL version 3 and the Affero General Public License. It is intended for software designed to be run over a network, adding a provision requiring that the corresponding source code of modified versions of the software be prominently offered to all users who interact with the software over a network. The Open Source Initiative approved the GNU AGPLv3 as an open source license in March 2008 after the company Funambol submitted it for consideration through its CEO Fabrizio Capobianco.

Now the Free Software Foundation has a legal team for people who breech the license and AnyCubic are currently in breech of the AGPL license they are using. They are _supposed_ to publish the modified source exactly for the reasons we are talking about here. If it were MIT or Apache or BSD license based this would be different, but OrcaSlicer (and its origins) were purposely using GPL based licensing for keeping changes and products "free and open".

tolistim, 2025-04-17:
The same with my Kobra 3s.  For now, I run the Windows version using a Windows 10 VM in VirtManager.