ANYCUBIC-3D/AnycubicSlicer #23: Connecting to Kobra S1 in lan mode is not working

angry-elf, 2025-03-08:
I would like to use LAN mode for my kobra s1 printer.

I enabled lan mode and start printer discovery in Anycubic SlicerNext program, but it can't find anything.

Both PC and printer are in the same network, i can ping it.

Official manual says, I should connect support in this situation.

yykato-org, 2025-03-29:
I am experiencing the same issue.

It is the same whether I use the version that was on the USB (size 99,426KB) or the latest one (size 100,862KB).

ks00x, 2025-04-05:
same for me

yykato-org, 2025-04-06:
> I am experiencing the same issue.
> 
> It is the same whether I use the version that was on the USB (size 99,426KB) or the latest one (size 100,862KB).

I am now able to connect.

However, I don't know what I did to enable the connection.
I think it was because I changed a setting somewhere, but I don't know which.
I'm not sure, but I think it might have been something to do with the LAN printer or cloud printer.

mental72, 2025-07-05:
Just making sure you all know to connect through Lan you need to connected with a cable, a printer cable also called a USB-B cable but i found out today that some versions of the Kobra S1 ship without a USB-B Port so therefore can only connect through WLAN  or wireless

IgooGG, 2026-10-11:
i had a same problem due to rinkhals. but once you install discovery helper it will fix it self.