ANYCUBIC-3D/AnycubicSlicer #28: Anycubic Slicer Next - Linux version

outerSpaceToast, 2025-05-01:
I am currently considering to buy a new printer. Either Anycubic Kobra S1 or Bambu P1S. However, it seems that there is no Linux version for the corresponding slicer.
Orca can be run under Linux, so please also provide a Linux version of Anycubic slicer or the option to use it fully with Orca slicer.

- Using Windows/MacOS is no option. 
- Having to use a USB flash drive to print is no option.
- Tinkering with a different firmware and being isolated from regular Anycubic updates is no option for me.

In case there is no way to use the Anycubic Kobra S1 regularly under Linux, I am forced to buy a Bambu Lab printer instead.

felix-ebbe, 2025-05-31:
Hi @takanuva15, yes Rinkhals is an option, but Anycubic Slicer Next is only a slightly modified version of Orca. Orca builds for Linux are available, so I cannot see a reason for Anycubic to not provide a Linux version as well. Yes, it does require a bit of additional testing, but the overall time usage is not that great. Another solution would be to make it possible to connect the original firmware to the main Orca version.

erdorath, 2025-06-23:
This issues need to be resolved by anycubic... the source code is not available (which under the GLP license it should be) so we cannot build ourselves for linux, the connection to the printer is not available in other slicers (orca, etc), we need more liberty to use the printer other than being forced to use a pre-compiled binary on windows or mac! Anycubic needs to open the connection to other slicers (by use of a plugin, for example) and release the code of which they are infringing the license! We mustn't have to rely on third parties creating compatible firmware and jailbreaks to properly use our printers...