ANYCUBIC-3D/AnycubicSlicer #36: AUTO Filament selction on the "REMOTE PRINT" page.

cdvankammen, 2025-07-23:
Please make it so that a user, under the settings tab or anything, is allowed to disable to "Switch filament" automatically to choose the best option. This provies me MANY MANY problems and having to restart prints halfway through. 

Most people I know do not stick to just AC filament and this casues so many issues as maybe I have not always chosen the specific color from the menu and taken the time to update that EVERY TIME i am wanting to print. BUT I do know the slot I want to use is #2 for example. When I am working with multiple machiens I know what color is load, I know what I want to do, but AC Next keeps changint it and I have to put more effort in EVERY SINGLE TIME to make sure it has not changed the slot to another filament color that may not even be loaded into that spot any more. 

RFIDs are in many filaments but you do not read their data....and I DO NOT always use AC filament. SO my colors will not always be correct. PLEASE PLEASE change this

TheToff, 2025-07-24:
I believe this could be resolved if synch was enabled between the three possible influences. The printer menu, the mobile App, and the Slicer. Filament types and their colour ought to be synched.

christophervankammen, 2025-07-24:
I believe this would be a good add on to the way to deal with this. Well at the same time. I believe that we should be able to just choose the filament I want with out needing to sync every single time. It takes a lot of extra effort when you have alot of machines to try and do this with every machine. 

It is more stressful having to worry okay is it going to do it this time. 

Also, with multiple machines I could be using the same project for multiple machines using multiple different colors on each machine. 

If there is just a simple ADVANCE MENU option even that allows "experienced users to just move things around as need or even allow a checkbox option" for change filament automatically. I just usually know that I'm looking at my ACE and place the filament in the same position slot on everyone of them